MMM-lab/TUKUBA_2021 #21: combine two 2D LiDAR scan into one

Ramune6110, 2020-09-26:
二つのlaser scan(複数のlaser scan)を一つに統合する

以下のパッケージが使えるかも

https://github.com/danieldugas/laser_scan_combiner

Ramune6110, 2020-10-02:
参考になりそうな資料

https://www.theconstructsim.com/ros-qa-how-to-merge-data-from-two-different-lasers/

https://medium.com/@amritgupta1999/merging-data-from-multiple-lidar-s-in-ros-e890fb60cbbf

ju-mingyue, 2021-06-25:
Hello，Ramune ， I would like to ask whether the two lidars that have been converted by the ira_laser_tools tool are also activated together when building maps? At present, I have encountered a problem. When two laser signals are running together when building a map, the signals will drift, but when I choose a laser radar, the tf coordinates will be inaccurate.
At the same time, when I run two lidars, obstacles cannot be added to the local map of autonomous navigation, so the obstacle avoidance function cannot be realized.